nxhack/openwrt-node-packages #1618: 17014 build error

hzjnet, 2023-06-21:
env: 
 lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux trixie/sid
Release:        n/a
Codename:       trixie   
this is debian 10.13

 gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 9.5.0-3' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-pHbIoc/gcc-9-9.5.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
gcc version 9.5.0 (Debian 9.5.0-3) 

openssl:
PKG_NAME:=openssl
PKG_BASE:=1.1.1
PKG_BUGFIX:=q
PKG_VERSION:=$(PKG_BASE)$(PKG_BUGFIX)
PKG_RELEASE:=1
PKG_USE_MIPS16:=0
ENGINES_DIR=engines-1.1

log :
/openwrt/lede-17.01.4-Mod-202306180100-opensslq/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/usr/include/openssl/ossl_typ.h:92:16: note: forward declaration of 'EVP_MD_CTX {aka struct evp_md_ctx_st}'
 typedef struct evp_md_ctx_st EVP_MD_CTX;
                ^~~~~~~~~~~~~
In file included from ../src/node.cc:16:0:
../src/node_crypto.h: In destructor 'virtual node::crypto::Hash::~Hash()':
../src/node_crypto.h:514:5: error: 'EVP_MD_CTX_cleanup' was not declared in this scope
     EVP_MD_CTX_cleanup(&mdctx_);
     ^~~~~~~~~~~~~~~~~~
../src/node_crypto.h:514:5: note: suggested alternative: 'EVP_MD_CTX_create'
     EVP_MD_CTX_cleanup(&mdctx_);
     ^~~~~~~~~~~~~~~~~~
     EVP_MD_CTX_create
../src/node_crypto.h: At global scope:
../src/node_crypto.h:567:14: error: field 'mdctx_' has incomplete type 'EVP_MD_CTX {aka evp_md_ctx_st}'
   EVP_MD_CTX mdctx_; /* coverity[member_decl] */
              ^~~~~~
In file included from /openwrt/lede-17.01.4-Mod-202306180100-opensslq/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/usr/include/openssl/crypto.h:25:0,
                 from /openwrt/lede-17.01.4-Mod-202306180100-opensslq/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/usr/include/openssl/comp.h:16,
                 from /openwrt/lede-17.01.4-Mod-202306180100-opensslq/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/usr/include/openssl/ssl.h:17,
                 from ../src/node_crypto.h:20,
                 from ../src/node.cc:16:
/openwrt/lede-17.01.4-Mod-202306180100-opensslq/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/usr/include/openssl/ossl_typ.h:92:16: note: forward declaration of 'EVP_MD_CTX {aka struct evp_md_ctx_st}'
 typedef struct evp_md_ctx_st EVP_MD_CTX;
                ^~~~~~~~~~~~~
In file included from ../src/node.cc:16:0:
../src/node_crypto.h: In destructor 'virtual node::crypto::SignBase::~SignBase()':
../src/node_crypto.h:561:5: error: 'EVP_MD_CTX_cleanup' was not declared in this scope
     EVP_MD_CTX_cleanup(&mdctx_);
     ^~~~~~~~~~~~~~~~~~
../src/node_crypto.h:561:5: note: suggested alternative: 'EVP_MD_CTX_create'
     EVP_MD_CTX_cleanup(&mdctx_);
     ^~~~~~~~~~~~~~~~~~
     EVP_MD_CTX_create
make[5]: *** [node.target.mk:151: /openwrt/lede-17.01.4-Mod-202306180100-opensslq/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/node-v4.4.5/out/Release/obj.target/node/src/node.o] Error 1
make[5]: Leaving directory '/openwrt/lede-17.01.4-Mod-202306180100-opensslq/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/node-v4.4.5/out'
make[4]: *** [Makefile:45: node] Error 2
make[4]: Leaving directory '/openwrt/lede-17.01.4-Mod-202306180100-opensslq/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/node-v4.4.5'
make[3]: *** [Makefile:95: /openwrt/lede-17.01.4-Mod-202306180100-opensslq/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/node-v4.4.5/.built] Error 2
make[3]: Leaving directory '/openwrt/lede-17.01.4-Mod-202306180100-opensslq/feeds/packages/lang/node'
make[2]: *** [package/Makefile:109: package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/openwrt/lede-17.01.4-Mod-202306180100-opensslq'
make[1]: *** [package/Makefile:104: /openwrt/lede-17.01.4-Mod-202306180100-opensslq/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/stamp/.package_compile] Error 2
make: *** [/openwrt/lede-17.01.4-Mod-202306180100-opensslq/include/toplevel.mk:200: world] Error 2

can you help me?   thanks

nxhack, 2023-06-21:
https://github.com/nxhack/openwrt-node-packages/issues/1613#issuecomment-1597876928

closed 2023-06-21 by nxhack

hzjnet, 2023-06-25:
> @hzjnet
> 
> Read Back Issues.
> 
> > 41 | #include <bits/c++config.h>
> 
> ```
> sudo apt-get install gcc-multilib g++-multilib
> ```

thank you  ,i finish it node 8.17

nxhack, 2023-06-25:
Hi @hzjnet 

This may be an unnecessary intervention, but I think it is the more correct judgment to make the effort to build with a safe version (OpenWrt-22.03).

hzjnet, 2023-06-26:
> Hi @hzjnet
> 
> This may be an unnecessary intervention, but I think it is the more correct judgment to make the effort to build with a safe version (OpenWrt-22.03).

maybe (~~)