nxhack/openwrt-node-packages #1726: Can't connect to Homebridge UI

lorenzofiamingo, 2023-09-19:
Hi,
I was trying to get Homebridge running on my Mi Router AX32000. I just installed `node-homebridge` and those where the logs:

```
Tue Sep 19 10:29:27 2023 daemon.err homebridge[2417]: [9/19/2023, 8:29:27 AM] config.json (/usr/share/homebridge/config.json) not found.
Tue Sep 19 10:29:31 2023 daemon.err homebridge[2417]: [9/19/2023, 8:29:31 AM] No plugins found. See the README for information on installing plugins.
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: Setup Payload:
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: X-HM://***
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: Scan this code with your HomeKit app on your iOS device to pair with Homebridge:
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:  ***
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: Or enter this code with your HomeKit app on your iOS device to pair with Homebridge:
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:     ┌────────────┐
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:      │ ***      │
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:     └────────────┘
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: [9/19/2023, 8:29:31 AM] Homebridge v1.2.4 is running on port 39609.
```

But when I connect to `192.168.1.1:39609` or `192.168.1.1:8581` I don't see the Homebridge Panel UI. Could someone please tell me if I'm missing any step?

nxhack, 2023-09-19:
The homebridge by itself does not have a web UI.

> Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: Scan this code with your HomeKit app on your iOS device to pair with Homebridge:
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]:  ***
Tue Sep 19 10:29:31 2023 daemon.info homebridge[2417]: Or enter this code with your HomeKit app on your iOS device to pair with Homebridge:

Scan the QR code displayed in the log with Home.app on your iPhone.

https://github.com/homebridge/homebridge/wiki/Connecting-Homebridge-To-HomeKit

lorenzofiamingo, 2023-09-20:
Thanks! My setup should be good as is then

closed 2023-09-20 by lorenzofiamingo