nxhack/openwrt-node-packages #1952: Node-Red on Raspberry pi 4 GPIO & Serial Port Issue

Quicksilverzone, 2024-02-26:
Hi,

I prepared openwrt image including node packets in these repo. But I cannot control GPIO and serial ports over Node-Red, it gives error on debug windows when I deploy the script on Node-Red with the gpio or serial node.  Is there any other point I need to change or configure I dont know. I am using raspberry pi CM4. I am a little bit newbie on openwrt platform. I'll be grateful for any help. Thank you.

[image]

nxhack, 2024-02-26:
Hi @Quicksilverzone 

Sorry, but I don't know much about it and this may give you a hint.
https://nodered.org/docs/faq/interacting-with-pi-gpio

https://www.raspberrypi.com/documentation/computers/configuration.html#serial

https://www.raspberrypi.com/documentation/computers/configuration.html#change-the-default-pin-configuration

nxhack, 2024-02-26:
I saw the error message in the image.
Try running this on the target machine.
```shell
npm install raspi-io
```

https://www.npmjs.com/package/raspi-io

nxhack, 2024-02-26:
https://openwrt.org/docs/techref/hardware/port.gpio

https://openwrt.org/docs/techref/hardware/port.serial