nxhack/openwrt-node-packages #1996: Please add homebridge-camera-ui

Albert711, 2024-05-16:
https://www.npmjs.com/package/homebridge-camera-ui
I will be very grateful.

nxhack, 2024-05-17:
Hi @Albert711 

I tried to port it, but there are some vulnerabilities in the modules that the "camera.ui" module depends on, and there is no fixed version, so I have to abandon the porting. Sorry for the inconvenience.

```
# npm audit report

dicer  *
Severity: high
Crash in HeaderParser in dicer - https://github.com/advisories/GHSA-wm7h-9275-46v2
No fix available
node_modules/dicer
  busboy  <=0.3.1
  Depends on vulnerable versions of dicer
  node_modules/busboy
    multer  <=2.0.0-rc.3
    Depends on vulnerable versions of busboy
    node_modules/multer

jsonwebtoken  <=8.5.1
Severity: moderate
jsonwebtoken unrestricted key type could lead to legacy keys usage  - https://github.com/advisories/GHSA-8cf7-32gw-wr33
jsonwebtoken's insecure implementation of key retrieval function could lead to Forgeable Public/Private Tokens from RSA to HMAC - https://github.com/advisories/GHSA-hjrf-2m68-5959
jsonwebtoken vulnerable to signature validation bypass due to insecure default algorithm in jwt.verify() - https://github.com/advisories/GHSA-qwph-4952-7xr6
No fix available
node_modules/jsonwebtoken
  socketio-jwt  *
  Depends on vulnerable versions of jsonwebtoken
  node_modules/socketio-jwt
```

Albert711, 2024-05-17:
But why not use multer@2.0.0-rc.4 instead of multer@2.0.0-rc.3, which no longer depends on the busboy, which in turn depends on the vulnerable dicer? Even busboy  >=1.0.0 didn't depends on the dicer anymore.

Albert711, 2024-05-17:
And the latest version of "camera.ui" (1.1.17) requires the "multer" module higher than 1.4.4 which, as I mentioned earlier, does not depend on the vulnerable modules. 
Maybe you trying to port some very old version of "camera.ui" module.

Albert711, 2024-05-17:
As for the "dicer" module, I figured out what was wrong with the audit failure.
It turns out that the "camera.ui" module uses the "multer" module version 1.4.4 instead of 1.4.4-lts.1 (I even tried 1.4.5-lts.1). After I changed the “camera.ui” module dependencies to the correct version, there are no more NPM audit warnings regarding the “dicer” module. Just need to add the right version to the[ patch file](https://github.com/nxhack/openwrt-node-packages/blob/212d6cbee2109e40fd25e0953f9b71ef0e9e6147/node-camera.ui/patches/000-remove_depends.patch).

But regarding the vulnerability in the "jsonwebtoken" module, here is the problem:
The "camera.ui" module uses the "socketio-jwt" module, which has not been updated for over 4 years (seems to be abandoned) and it, in turn, uses an old version of the "jsonwebtoken" module (with a vulnerability).
Could you add the "socketio-jwt" module to your repository and patch it to use the latest version of the "jsonwebtoken" module? And now we can use this for the “camera.ui” module.

And I also noticed that in one of the modules of your repository, [“tar” was removed](https://github.com/nxhack/openwrt-node-packages/blob/master/node-ffmpeg-for-homebridge/patches/000-remove_depends.patch) from the module dependencies. Maybe this also needs to be removed from the dependencies of the "camera.ui" module? I just don't understand this very deeply.

nxhack, 2024-05-18:
"camara.ui" gives this warning. In any case, it is not desirable because it is not maintained.

```
npm warn deprecated gar@1.0.4: Package no longer supported. Contact Support at https://www.npmjs.com/support for more info.
npm warn deprecated har-validator@5.1.5: this library is no longer supported
npm warn deprecated querystring@0.2.1: The querystring API is considered Legacy. new code should use the URLSearchParams API instead.
npm warn deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm warn deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm warn deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
```

Albert711, 2024-05-18:
Well, let's see if I can at least do something about it.

nxhack, 2024-05-19:
For now. I was able to adjust the module version consistency and start it with config-ui. Please check the actual connection to the camera device.

Albert711, 2024-05-19:
Thank you @nxhack 
Today I will definitely try to check everything.

Albert711, 2024-07-14:
> For now. I was able to adjust the module version consistency and start it with config-ui. Please check the actual connection to the camera device.

Albert711, 2024-07-14:
Thank you for your responsiveness and for the quick implementation. I am grateful to you.

And I apologize for such a late reply. The household routine bogged me down completely.

I checked everything and I have to say that everything works as expected. I am satisfied.

And of course it is a pity that the author of the plugin has not updated this for a long time. During this time I also managed to test [Scrypted](https://github.com/koush/scrypted). And I must say it turned out to be a little better. But unfortunately (as far as I understand), it is impossible to integrate this into OpenWrt. And it requires an additional server, which is not always convenient and possible. I hope the author of the camera.ui plugin will wake up and fix all the imperfections.

Albert711, 2024-08-07:
@nxhack  Is it possible to also add [Scrypted](https://www.npmjs.com/package/scrypted)?

nxhack, 2024-08-08:
scrypted seems to be a huge, complex application server.
It seems that simply 'npm install scrypted' is not enough, so I can't do anything about it.

I introduced 'scrypted' in my local environment to try it out, but with require('scrypted');,
```
> require('scrypted');
usage:
   npx scrypted install npm-package-name [127.0.0.1[:10443]]
   npx scrypted install npm-package-name[/0.0.1] [127.0.0.1[:10443]]
   npx scrypted login [127.0.0.1[:10443]]
   npx scrypted serve
   npx scrypted serve@latest
   npx scrypted serve[@version]
   npx scrypted command name-or-id[@127.0.0.1[:10443]] method-name [...method-arguments]
   npx scrypted ffplay name-or-id[@127.0.0.1[:10443]] method-name [...method-arguments]
   npx scrypted create-cert-json /path/to/key.pem /path/to/cert.pem
   npx scrypted shell [127.0.0.1[:10443]] [-- cmd [...cmd-args]]

examples:
   npx scrypted install @scrypted/rtsp
   npx scrypted install @scrypted/rtsp/0.0.51
   npx scrypted install @scrypted/rtsp/0.0.51 192.168.2.100
{}
> 
```

I don't have the time and energy to cover everything.