nxhack/openwrt-node-packages #2017: node-vite

esaaprillia, 2024-06-13:
hy @nxhack 

i build node-vite but get error. can you add node-vite?

node-v20.14.0

npm error code EUNSUPPORTEDPROTOCOL
npm error Unsupported URL Type "link:": link:./src/types

nxhack, 2024-06-13:
Hi @esaaprillia 

Can you give me the link to npm?

esaaprillia, 2024-06-13:
> Hi @esaaprillia
> 
> Can you give me the link to npm?

https://www.npmjs.com/package/vite

esaaprillia, 2024-06-13:
```
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NPM_NAME:=vite
PKG_NAME:=node-$(PKG_NPM_NAME)
PKG_VERSION:=5.2.13
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NPM_NAME)-$(PKG_VERSION).tgz
PKG_SOURCE_URL:=https://registry.npmjs.org/$(PKG_NPM_NAME)/-/
PKG_HASH:=314792d2c86a216451f8abc4a992f8e9c0b1f1963ce77a6ec8b93d628bc25508

PKG_MAINTAINER:=Hirokazu MORIKAWA <morikw2@gmail.com>
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE

HOST_BUILD_DEPENDS:=node/host
HOST_BUILD_PARALLEL:=1

PKG_BUILD_DEPENDS:=node/host
PKG_BUILD_PARALLEL:=1
PKG_BUILD_FLAGS:=no-mips16

include $(INCLUDE_DIR)/host-build.mk
include $(INCLUDE_DIR)/package.mk

define Package/node-vite
  SUBMENU:=Node.js
  SECTION:=lang
  CATEGORY:=Languages
  TITLE:=Next Generation Frontend Tooling
  URL:=https://www.npmjs.com/package/vite
  DEPENDS:=+node
endef

define Package/node-vite/description
 Vite (French word for "quick", pronounced /vit/, like "veet") is a new breed of frontend build tooling that significantly improves the frontend development experience.
endef

TAR_OPTIONS+= --strip-components 1
TAR_CMD=$(HOST_TAR) -C $(1) $(TAR_OPTIONS)

NODEJS_CPU:=$(subst powerpc,ppc,$(subst aarch64,arm64,$(subst x86_64,x64,$(subst i386,ia32,$(ARCH)))))
TMPNPM:=$(shell mktemp -u XXXXXXXXXX)
HOSTTMPNPM:=$(shell mktemp -u XXXXXXXXXX)
NPM_CACHE_DIR:=$(if $(call qstrip,$(CONFIG_NODEJS_NPM_CACHE_DIR)),$(call qstrip,$(CONFIG_NODEJS_NPM_CACHE_DIR)),$(TMP_DIR))

TARGET_CFLAGS+=$(FPIC)
TARGET_CPPFLAGS+=$(FPIC)

define Build/Compile
	cd $(PKG_BUILD_DIR); \
	$(MAKE_VARS) \
	$(MAKE_FLAGS) \
	npm_config_nodedir=$(STAGING_DIR)/usr/ \
	$(if $(CONFIG_NODEJS_NPM_KEEP_CACHE), npm_config_cache=$(NPM_CACHE_DIR)/npm-cache-$(PKG_NPM_NAME),npm_config_cache=$(TMP_DIR)/npm-cache-$(TMPNPM)) \
	npm install --prefer-offline --no-audit --global-style --install-strategy=shallow --no-save --omit=dev --no-package-lock --ignore-scripts
	rm -rf $(TMP_DIR)/npm-cache-$(TMPNPM)
	rm -f $(PKG_BUILD_DIR)/node_modules/.package-lock.json
	find $(PKG_BUILD_DIR)/node_modules -type d -empty -print0 | xargs -0 rmdir || true
endef

define Package/node-vite/install
	$(INSTALL_DIR) $(1)/usr/lib/node/$(PKG_NPM_NAME)
	$(CP) $(PKG_BUILD_DIR)/{package.json,*.cts,*.cjs,*.ts,*.md} \
		$(1)/usr/lib/node/$(PKG_NPM_NAME)/
	$(CP) $(PKG_BUILD_DIR)/{bin,dist,types,node_modules} \
		$(1)/usr/lib/node/$(PKG_NPM_NAME)/
	$(INSTALL_DIR) $(1)/usr/bin
	$(LN) ../lib/node/$(PKG_NPM_NAME)/bin/vite.js $(1)/usr/bin/vite
	$(INSTALL_DIR) $(1)/usr/lib/node_modules
	$(LN) ../node/$(PKG_NPM_NAME) $(1)/usr/lib/node_modules/$(PKG_NPM_NAME)
endef

define Package/node-vite/postrm
#!/bin/sh
rm /usr/lib/node_modules/vite || true
rm -rf /usr/lib/node/vite || true
endef

define Host/Compile
	cd $(HOST_BUILD_DIR); \
	$(HOST_MAKE_VARS) \
	npm_config_nodedir=$(STAGING_DIR)/usr/ \
	$(if $(CONFIG_NODEJS_NPM_KEEP_CACHE), npm_config_cache=$(NPM_CACHE_DIR)/host-npm-cache-$(PKG_NPM_NAME),npm_config_cache=$(TMP_DIR)/npm-cache-$(HOSTTMPNPM)) \
	npm install --prefer-offline --no-audit --global-style --install-strategy=shallow --no-save --omit=dev --no-package-lock --ignore-scripts
	rm -rf $(TMP_DIR)/npm-cache-$(HOSTTMPNPM)
	rm -f $(HOST_BUILD_DIR)/node_modules/.package-lock.json
	find $(HOST_BUILD_DIR)/node_modules -type d -empty -print0 | xargs -0 rmdir || true
endef

define Host/Install
	$(INSTALL_DIR) $(1)/lib/node_modules/$(PKG_NPM_NAME)
	$(CP) $(HOST_BUILD_DIR)/{package.json,*.cts,*.cjs,*.ts,*.md} \
		$(1)/lib/node_modules/$(PKG_NPM_NAME)/
	$(CP) $(HOST_BUILD_DIR)/{bin,dist,types,node_modules} \
		$(1)/lib/node_modules/$(PKG_NPM_NAME)/
	$(INSTALL_DIR) $(1)/bin
	$(LN) ../lib/node_modules/$(PKG_NPM_NAME)/bin/vite.js $(1)/bin/vite
endef

$(eval $(call HostBuild))
$(eval $(call BuildPackage,node-vite))

```

nxhack, 2024-06-13:
I have made the build succeed, but it looks like I probably need to work on compiling the target binaries for esbuild and rollup. This may take some time.

https://github.com/nxhack/openwrt-node-packages/tree/dev/node-vite

```
  "dependencies": {
    "esbuild": "^0.20.1",
    "postcss": "^8.4.38",
    "rollup": "^4.13.0"
  },
```

esaaprillia, 2024-06-13:
> I have made the build succeed, but it looks like I probably need to work on compiling the target binaries for esbuild and rollup. This may take some time. [...]
Thank you. do we need to build a rollup plugin

```
package.json:    "rollup": "^4.13.0"                                                    
package.json:    "@rollup/plugin-alias": "^5.1.0",                                      
package.json:    "@rollup/plugin-commonjs": "^25.0.8",                                  
package.json:    "@rollup/plugin-dynamic-import-vars": "^2.1.2",                        
package.json:    "@rollup/plugin-json": "^6.1.0",                                       
package.json:    "@rollup/plugin-node-resolve": "15.2.3",                               
package.json:    "@rollup/plugin-typescript": "^11.1.6",                                
package.json:    "@rollup/pluginutils": "^5.1.0",                                       
package.json:    "rollup-plugin-dts": "^6.1.1",                                         
package.json:    "rollup-plugin-esbuild": "^6.1.1",                                     
package.json:    "rollup-plugin-license": "^3.4.0",
```

nxhack, 2024-06-13:
It seems to work even without the esbuild and rollup binary modules.
```
> require('vite');
The CJS build of Vite's Node API is deprecated. See https://vitejs.dev/guide/troubleshooting.html#vite-cjs-node-api-deprecated for more details.
{
  defineConfig: [Function (anonymous)],
  esbuildVersion: '0.20.2',
  createFilter: [Function: createFilter],
  createLogger: [Function: createLogger],
  isCSSRequest: [Function: isCSSRequest],
  isFileServingAllowed: [Function: isFileServingAllowed],
  loadEnv: [Function: loadEnv],
  mergeAlias: [Function: mergeAlias],
  mergeConfig: [Function: mergeConfig],
  normalizePath: [Function: normalizePath],
  resolveEnvPrefix: [Function: resolveEnvPrefix],
  rollupVersion: '4.18.0',
  searchForWorkspaceRoot: [Function: searchForWorkspaceRoot],
  send: [Function: send],
  splitVendorChunk: [Function: splitVendorChunk],
  splitVendorChunkPlugin: [Function: splitVendorChunkPlugin],
  version: '5.2.13',
  build: [Function (anonymous)],
  createServer: [Function (anonymous)],
  preview: [Function (anonymous)],
  transformWithEsbuild: [Function (anonymous)],
  resolveConfig: [Function (anonymous)],
  optimizeDeps: [Function (anonymous)],
  formatPostcssSourceMap: [Function (anonymous)],
  loadConfigFromFile: [Function (anonymous)],
  preprocessCSS: [Function (anonymous)]
}
> 
```

esaaprillia, 2024-06-13:
> It seems to work even without the esbuild and rollup binary modules.
>
> ```
> The CJS build of Vite's Node API is deprecated. See [...]

Good. can we add experiment packages node-telegram-tt and node-tweb? I want to test it

https://github.com/Ajaxy/telegram-tt

https://github.com/morethanwords/tweb

esaaprillia, 2024-06-13:
@nxhack 

can you see this

https://github.com/vitejs/vite/issues/17456