nxhack/openwrt-node-packages #2045: Please add homebridge-ewelink

Albert711, 2024-07-14:
One more small request 😊
Very useful plugin for me. And this time the plugin author regularly updates it.

https://www.npmjs.com/package/homebridge-ewelink

nxhack, 2024-07-16:
I have confirmed that it works with the homebridge web UI.
Please test. 

https://github.com/nxhack/openwrt-node-packages/tree/master/node-homebridge-ewelink
https://github.com/nxhack/openwrt-node-packages/tree/master/node-homebridge-miot

Albert711, 2024-07-18:
I tested. The plugins work properly.
There is only one small cosmetic remark. Here's no plugin description:
[image]